stprograms/SuperSoco485Monitor #1: not full protocol

pervolianinen, 2023-05-09:
in version 0.9 SuSoDevs new information. data from ecu (gps/gsm) module. I ask Xmanu12 for project 0.9, but he didn't  give it.

stprograms, 2023-05-10:
Hi @pervolianinen !
At the moment, this is all the information I could extract from the provided kodular project file and the discussion on the project. Now that you say it, the checked in kodular project is of version 0.4 and the current version is 0.9. Since I could not find any additional information on GPS info, I assumed that the GPS information was taken from the android application.

Non the less, the SuperSoco485 Monitor can be used to further analyse the traffic and maybe extract the required info. I'm also planning on a different display to only show the last data per received telegram

pervolianinen, 2023-05-19:
a5 0a 20 ff 20 22 f7 20 20 20 0b 8d f7 d4  
a5 0a 20 ff 20 22 f8 20 20 20 0b 8d f8 d4  
a5 0a 20 ff 20 22 f9 20 20 20 0b 8d f9 d4  

7
[ecu.log](https://github.com/stprograms/SuperSoco485Monitor/files/11518969/ecu.log)
 block - seconds (f7, f8, f9...)

a5 0a 20 ff 20 77 bf 20 20 20 0b e2 bf ee

stprograms, 2023-05-23:
Hi!
What do you mean with micro USB port? Does the ECU have a port where you can directly connect yourself to it?

Also thanks for the log. Sadly, I do not see anything that matches the protocol pattern. Can you try to switch the two lines of the RS485? During my first tries, I had a similar pattern of the raw data and there I had switched the lines.

pervolianinen, 2023-06-04:
> What do you mean with micro USB port? Does the ECU have a port where you can directly connect yourself to it?

yes. ECU has power/data port.

1 file ecu work bike blocked
[serial_20230604_183039.txt](https://github.com/stprograms/SuperSoco485Monitor/files/11646155/serial_20230604_183039.txt)

bike blocked signalization work (sound)
[serial_20230604_183210.txt](https://github.com/stprograms/SuperSoco485Monitor/files/11646156/serial_20230604_183210.txt)

bike unblocked 
[serial_20230604_183237.txt](https://github.com/stprograms/SuperSoco485Monitor/files/11646157/serial_20230604_183237.txt)

stprograms, 2023-06-27:
Hi!
Thanks for the output and sorry for the late response.
I finally got some time to look into the output, but sadly could not find any new information, but it looks similar to the trace of my TCMax. My assumption is that the ECU (where the app connects to), does not send the GPS information an the bus. It does not need to, since it has the GPS information itself and also the information for the (GSM) network.

I have also released the latest version of the monitor 1.1.0 that now supports live decoding of the received data. That should allow you to see changes live (eg. parking mode or time).

The raw data can also be stored directly into a file, if you specify so in the appsettings.json. This allows recording and an "offline" replay and analysis later :)

closed 2023-08-08 by stprograms