stprograms/SuperSoco485Monitor #2: aboute docuentation

pervolianinen, 2023-07-15:
There is error in documentation.
This project will be work with all actual supers soco.
All super soco have standart protocol, but use 2 transport protocols CAN and RS485.
Android app SUSODEV work with all super soco. in app no information about transport protocol CAN or RS485

stprograms, 2023-08-08:
Hi pervolianinen!

You are right, the Android app from Susodev works with both types of motorcycles. Those that use RS485 and CAN.
**BUT**: If dive further into the documentation of SuSoDevs work and the project itself you will see two things:

The motorcycles with RS485 use a RS485 to BT hardware, the ones with CAN use a converter from CAN to BT. This interface holds an ESP32 loaded with a *sketch* that **translates** the CAN messages to the protocol used on RS485 and transmits it over BT, so the application gets data it can handle.

So yes, if you build the CAN adapter and use the BT serial on the PC as the serial interface for the monitor, you should also be able to read the data from newer models using CAN.

CAN and the RS485 interface of SuperSoco are two completely different interfaces and use different protocols. A good comparison would be VOIP and morse code.

If you like to fetch more information on the Super Soco CAN interface, you can check out [Volt Garaz Wiki](https://wiki.voltgaraz.cz/index.php/CAN) where the messages are already documented.

Of course you are welcome to document the CAN protocol in the same way as I did and add it to the README.md file, so we have one place where all the information is stored :) The monitor can in theory be extended to support CAN interfaces too.

pervolianinen, 2023-08-13:
> Hi pervolianinen!
> 
> You are right, the Android app from Susodev works with both types of motorcycles. Those that use RS485 and CAN. _**BUT**_: If you dive further into the documentation of SuSoDevs work and the project itself you will see two things:
> 
> 1. The motorcycles with RS485 use a RS485 to BT hardware, the ones with CAN use a converter from CAN to BT.
> 2. This interface holds an ESP32 loaded with a _sketch_ that **translates** the CAN messages to the protocol used on RS485 and transmits it over BT, so the application gets data it can handle.
> 
> So yes, if you build the CAN adapter and use the BT serial on the PC as the serial interface for the monitor, you should also be able to read the data from newer models using CAN.
> 
> CAN and the RS485 interface of SuperSoco are two completely different interfaces and use different protocols. A good comparison would be VOIP and morse code.
> 
> If you like to fetch more information on the Super Soco CAN interface, you can check out [Volt Garaz Wiki](https://wiki.voltgaraz.cz/index.php/CAN) where the messages are already documented.
> 
> Of course you are welcome to document the CAN protocol in the same way as I did and add it to the README.md file, so we have one place where all the information is stored :) The monitor can in theory be extended to support CAN interfaces too.

You are not right.
The protocol for Lingbo controllers is the same for all mopeds Super Soco. Transport protocol may be CAN, RS485, UART.
For CAN, due to the specifics of the protocol, there are no ready-made converters to UART.
But Supers Soco protocol same in all.

stprograms, 2023-08-21:
You are definitely deeper into the details of the Super Soco vehicles than I am :) 

So, what you are saying is that the communication between the components is always the same protocol I have documented here and the monitor can be used for the RS485 interface and the eg. USB UART of the controller directly, correct?

CAN of course uses a different communication protocol and is not compatible here, but according to the code of SuSoDevs CANInterface, the data is just packed into the payload of the can message. 

So you're suggestion is to update the documentation to hold the hint, that it may be also used for newer bikes using CAN, correct, even if there needs to be a special interface that converts it?

pervolianinen, 2024-09-13:
BMS Status (Read Response 0xAA5A)

неизвестные байты 6 и 7 это количество разрядок.
[image]

pervolianinen, 2024-09-15:
For next request C5 5C 5A AA 01 9A 9B 0D (9A command)
Answer B6 6B AA 5A 02 0F 03 0E 0D (0F 03)

stprograms, 2024-10-13:
Hi @pervolianinen!
Sorry for the late reply, but I currently do not have much spare time and I'm also lacking the possibility to test the telegrams live. I will add the Discharge information to the documentation and the telegram parser.

stprograms, 2024-10-13:
@pervolianinen !
Could you please check out the beta release https://github.com/stprograms/SuperSoco485Monitor/releases/tag/v1.2.0-beta.1 and verify if the Discharging Cycles are parsed correctly? I currently do not have the setup to  check it.

If it works, I would merge it into master then

pervolianinen, 2024-10-13:
> @pervolianinen ! Could you please check out the beta release https://github.com/stprograms/SuperSoco485Monitor/releases/tag/v1.2.0-beta.1 and verify if the Discharging Cycles are parsed correctly? I currently do not have the setup to check it.
> 
> If it works, I would merge it into master then

there is no *.exe file in zip.

stprograms, 2024-10-13:
> there is no *.exe file in zip.

Thanks! I have updated the zip file. Please try again

PR #7 (2024-10-17): 2 discharge cycle information
# Changes
- Added discharging cycles to BMS telegram
- Added information from @pervolianinen regarding different interfaces

Waiting for verification by @pervolianinen

closes #2
closed 2024-10-17 by stprograms