stprograms/SuperSoco485Monitor #5: Monitoring/ comunication Protocol

Elias-Ta, 2024-03-27:
Thank you for explaining the communication protocol, I am trying to use Labview to monitor a super soco tc with a UI, however I don't quite understand how your software work, does it only listen to msgs in the RS485 network and translate these msgs, or it request specific data every time it's needed ?

What "Read Response 0xAA5A" means for example, which byte should be AA5A to recieve BMS status, and how to request this responce ? can you explain more please if you have time

stprograms, 2024-03-28:
Hi @Elias-Ta ,
Thank you for your response, I'm happy that I could help you.
Basically what I'm doing is only listening to the messages on the interface. If the bikes ignition is on, the BMS and ECU data is transmitted approximately every three seconds. If it is off, no messages from ECU are transmitted, but BMS seems it's status every 5 seconds or so if I remember correctly.
If you take a look at the discussions from the linked project, there seems to be the possibility to query the BMS actively, but that was not in my scope and I didn't test it.

If you just want to monitor the status, it is sufficient to decode only the messages that start with the first bytes specified in the tables.
Since there is no official documentation on the protocol and all the information here is based on reverse engineering, read response and so on is just a guess.

I hope I could help you a little bit. :-)
Good luck with you project

towe96, 2024-09-24:
> I get nnew data about protocol
> 
> C55C5AAA0197960D - standart qustion to bms in protocol
> 
> C55C5AAA0198980D - qustion to bms about voltage of each parallels.

Hello pervolianinen,
I'm currently building a new dashboard app for my TC Max after I've found the SuSoDevs app to be quite buggy.
I've tried out your frame (`C5 5C 5A AA 01 98 99 0D`, `99` because of the checksum calculation) and got a response from my BMS;
`B6 6B AA 5A 0A 4D 47 13 00 01 7A 00 8C 00 00 E4 0D`
The checksum (`E4`) is correct so I'm confident it's a complete frame; it's sometimes a bit difficult to tell for me since I'm only injecting commands into the live bus for now instead of talking to just the BMS.
Unfortunately I can't see a pattern in this data for now that would indicate cell voltages - maybe you'll have more luck? Mine is a 72V 7-Pin battery from a 2020 TC Max.

Is there any chance you could share the program in your screenshot? I haven't found anything like that yet.

pervolianinen, 2024-10-06:
Wright not correct information
C5 5C 5A AA 01 97 96 0D query data
C5 5C 5A AA 01 98 99 0D query voltage
C5 5C 5A AA 01 99 98 0D query version

97 96 
Answer

Voltage SoC Temp Charge CycleH CycleL DisgchargeCycleH DisgchargeCycleH VBreaker Charging

C5 5C 5A AA 01 9A 9B 0D question to bms from charger
Answer of not connected battery
B6 6B AA 5A 02 0F 03 0E 0D