stprograms/SuperSoco485Monitor #10: Some feedback

linshuweitw, 2024-12-06:
Hi @stprograms ,

I've recently been analyzing the communication protocol of Super Soco, and this project has been incredibly helpful to me.

Here are some of my findings and questions:

It’s known that the communication is based on a 2-wire RS485. From my tests, I’ve identified the following components in the system that actively send signals:
ECU
MCU (Motor Control Unit)
Speed Meter
Battery
The Charger should also participate, but I haven’t tested it yet.

Observations:
ECU: Acts as the Master and sends requests (REQ).
MCU: Responds with RESP(B6 6B AA DA....). I think referring to byte 5 as the "ECU state" is inaccurate; it might be better described as the "Drive state."
Battery: Responds with RESP(B6 6B AA 5A...).
Speed Meter: Responds with RESP(B6 6B AA BA....). Interestingly, if it doesn’t receive a REQ (C5 5C BA AA) for some time, it becomes the Master and starts sending REQ.
Additional Notes:
The interpretation of byte2 = src and byte3 = dest is incorrect.
It should be:
byte2 = dest
byte3 = src
From the above information, the address mappings are as follows:

Master Address = AA
MCU Address = DA
Battery Address = 5A
Speed Meter Address = BA

In the REQ (C5 5C BA AA...), the following observations are noted:
Byte 5: Represents the Battery State of Charge (SoC).
Byte 11: Contains an Error State. When the Battery is disconnected, this value becomes 20.

stprograms, 2024-12-08:
Hi @linshuweitw !
Thank you very much for your feedback. It is great to hear that the project was helpful to you and thank you very much for sharing your findings. From your description, it seems that you really have taken the bike completely apart when you can identify what component send what 😄 

For the charger, I know that there is some communication between the charger and the battery, according to [this thread](https://github.com/Xmanu12/SuSoDevs/issues/2). Maybe the information there can help you. 

Regarding your input, I will try to incorporate it, but it will take some time. If you would like to participate, you can create a branch, extend the readme.md and create a pull request. This way, we remove any possible misunderstandings from my side. I will then try to update the source code afterwards.

Please let me know if you want to contribute.

linshuweitw, 2024-12-25:
Hi @stprograms  , sorry for the delayed reply.

Yes, I disassembled a Super Soco and connected the communication lines for analysis.
I’ll continue working on the protocol and will update you with any new findings. I’m not very familiar with using GitHub, so I prefer to directly share my discoveries with you.
If you need my help, such as identifying the location of components on the bike or providing photos, I’d be more than happy to assist.

linshuweitw, 2025-01-08:
Hi @stprograms ,
The experiment has concluded, and the document is being edited. Some parts have already been updated, and you can take a look whenever you have time.

stprograms, 2025-01-08:
Thank  you very much for your input and congratulations that your experiment succeeded :) I also added you as a collaborator to this project.

I have created a new branch 10-documentation where I pulled your changes. It would be great if you could continue updating in this branch here, so we can merge them later to the main branch. I will also make some adaptions here later so it is easier to keep things consistent. Looking forward on working together

linshuweitw, 2025-01-13:
Hi, the document update has been completed. Let me know if there is an issue.

stprograms, 2025-01-17:
Thank you very much! 
I will copy and refine them when I have time :-)

stprograms, 2025-02-16:
Hi there!
I have now integrated your findings in a new branch and created a pull request. Could you please take a look on it? #15 
Thank you very much :)

linshuweitw, 2025-02-17:
Hi @stprograms , 
I will review when I have time.

closed 2025-02-18 by stprograms

valy84, 2025-03-09:
> Hi [@stprograms](https://github.com/stprograms) ,
>
> I've recently been analyzing the communication protocol of Super Soco, and this project has been incredibly helpful to me.
>
> Here are some of my [...]

Hello and thank you so much for sharing your findings. I have been using esp32s to send data to a SuperSoco TS1200r and TS Street Hunter Pro for more than an year now and I can emulate battery 1, battery 2 and controller without issues. However I don't have the original GSM (ECU) module and I am trying to emulate it and even with your latest findings I have no luck in doing so. I cannot make the esp32 act as a master. I try sending "C5 5C BA AA 0E 64 00 00 01 05 24 00 00 01 01 00 00 00 8C C6 0D" every 100,200,300 ms but no luck, the display takes command as master. Do you have any idea what I might be doing wrong? Thanks.

PR #15 (2025-02-18): Documentation update
This pull request incorporates the findings and changes of @linshuweitw . The documentation in the readme has been extended and formulated further.

closes #10