stprograms/SuperSoco485Monitor #27: > Hi [@stprograms](https://github.com/stprograms) ,

stprograms, 2025-03-11:
> Hi [@stprograms](https://github.com/stprograms) ,
> 
> I've recently been analyzing the communication protocol of Super Soco, and this project has been incredibly helpful to me.
> 
> Here are some of my findings and questions:
> 
> It’s known that the communication is based on a 2-wire RS485. From my tests, I’ve identified the following components in the system that actively send signals: ECU MCU (Motor Control Unit) Speed Meter Battery The Charger should also participate, but I haven’t tested it yet.
> 
> Observations: ECU: Acts as the Master and sends requests (REQ). MCU: Responds with RESP(B6 6B AA DA....). I think referring to byte 5 as the "ECU state" is inaccurate; it might be better described as the "Drive state." Battery: Responds with RESP(B6 6B AA 5A...). Speed Meter: Responds with RESP(B6 6B AA BA....). Interestingly, if it doesn’t receive a REQ (C5 5C BA AA) for some time, it becomes the Master and starts sending REQ. Additional Notes: The interpretation of byte2 = src and byte3 = dest is incorrect. It should be: byte2 = dest byte3 = src From the above information, the address mappings are as follows:
> 
> Master Address = AA MCU Address = DA Battery Address = 5A Speed Meter Address = BA
> 
> In the REQ (C5 5C BA AA...), the following observations are noted: Byte 5: Represents the Battery State of Charge (SoC). Byte 11: Contains an Error State. When the Battery is disconnected, this value becomes 20.

Hello and thank you so much for sharing your findings. I have been using esp32s to send data to a SuperSoco TS1200r and TS Street Hunter Pro for more than an year now and I can emulate battery 1, battery 2 and controller without issues. However I don't have the original GSM (ECU) module and I am trying to emulate it and even with your latest findings I have no luck in doing so. I cannot make the esp32 act as a master. I try sending "C5 5C BA AA 0E 64 00 00 01 05 24 00 00 01 01 00 00 00 8C C6 0D" every 100,200,300 ms but no luck, the display takes command as master. Do you have any idea what I might be doing wrong? Thanks.

_Ursprünglich gepostet von @valy84 in https://github.com/stprograms/SuperSoco485Monitor/issues/10#issuecomment-2708986587_

stprograms, 2025-03-11:
Hi @valy84!
I think emulating the original Ecu is an interesting project, good luck on that :) I have moved your question to a new issue.
Regarding your question, I cannot help you at the moment. I assume that either there is 1 special telegram sent that is not yet found / covered, or we have a special bit in a telegram of the Ecu that is not yet decoded.

@linshuweitw: Maybe you have an idea what makes the master role switch back to the Ecu

As you may have seen in the issues, I plan on adding timing to the traces, so the data communication can be analyzed and replayed with the same timing as it is recorded. Maybe I will share a recording of my bike when finished as an example

valy84, 2025-03-11:
Hi @stprograms and thanks for the reply. You putting all this information together has been super helpful as my Socos both run Fardriver controllers and JBD/JK BMS' and I am able to comunicate with the display and send all my data (well, most of it, except for the clock and real range). Regarding the ECU emulation, I even tried having the ESP running with the bike off, so that there is no delay in the sent packages when the display turns on, but the results are the same. I know it's not that big of a deal and probably emulating that is not of much interest to most people, but it would be nice to have 100% control on that display with full functionality. Thanks again!

harkiratsaggu10, 2026-09-19:
Hi everyone,

I’m hoping someone here can point me in the right direction. I own a Revolt RV400 (which I understand is based on the Super Soco platform) with a 72V lithium-ion battery. The BMS appears to have entered a permanent protection lockout and I’m stuck.

Symptoms:

· Battery was charging very slowly (up to 14 hours to reach 99%, then stopping).
· Eventually it stopped charging completely. The charger now runs cold (no current), and the bike is totally dead.
· No external damage, no overheating, always used the original charger.
· Service center says State of Health (SOH) is low and denied warranty. They want ~₹67,000 for a new battery, which I can’t afford.

What I’ve tried:

· Checked fuses, connections, charger output – all seem fine.
· Tried a basic BMS reset (disconnecting balance/main leads overnight) – no luck.
· Battery stays cool during charging attempts, so no obvious short or fire risk.
· I have not attempted any unsafe bypass or generic BMS swap.

What I’m looking for:

· Any information on the CAN bus / RS485 protocol used by Super Soco / Revolt BMS.
· Known diagnostic tools, software, or adapters (e.g., CANable, USB-RS485) that can read/clear BMS fault codes.
· Pinouts, firmware, or service manuals for the BMS.
· Experience from anyone who has successfully revived a locked Super Soco BMS, or replaced it with a compatible unit that still communicates with the motor controller.
· Pointers to any relevant GitHub repos, forum threads, or Discord/Telegram groups.

I’m based in India but can buy hardware online and follow remote guidance. I’m willing to provide logs, test, and contribute back whatever I learn.

If you’ve dealt with this before or know where I should look, I’d really appreciate your help.

Thanks!

BrienAllison, 2026-10-20:
Hi @valy84 and @stprograms . Maybe a little off topic but I don't know where else to ask this. I am just troubleshooting my TC Max bike using the RS485 monitor, and I believe my ECU is not communicating with the other components on the bike. I am only getting battery and controller requests/responses, nothing to or from the speedometer. I assume this is because the speedometer has taken the master role, not the ECU. Does this match what you saw from your bikes?

As a result my speedometer is not working, neither the battery percentage or range is shown on the display (but it does do a sweep when first powered on). Otherwise the bike works as expected. If I unplug the ECU unit the bike works the same (I have yet to monitor the RS485 bus with this configuration but I would assume the monitor output would be the same). If I unplug the speedo unit but leave the ECU connected, nothing on the bike works. The three LEDs light up on the ECU unit when connected, it just doesn't seem to be communicating.

I'm fairly certain this is an ECU issue, I just wanted to confirm if this is the case.

valy84, 2026-10-20:
> 6\. Hi [@valy84](https://github.com/valy84) and [@stprograms](https://github.com/stprograms) . Maybe a little off topic but I don't know where else to ask this. I am just troubleshooting my TC Max bike using the RS485 monitor, and I believe my ECU is not communicating with the other components on the bike. I am only getting battery and controller requests/responses, nothing to or from the speedometer. I assume this is because the speedometer has taken the master role, not the ECU. Does this match what you saw from your bikes?
>     As a result my speedometer is not working, neither the battery percentage or range is shown on the display (but it does do a sweep when first powered on). Otherwise the bike works as expected. If I unplug the ECU unit the bike works the same (I have yet to monitor the RS485 bus with this configuration but I would assume the monitor output would be the same). If I unplug the speedo unit but leave the ECU connected, nothing on the bike works. The three LEDs light up on the ECU unit when connected, it just doesn't seem to be communicating.
>     I'm fairly certain this is an ECU issue, I just wanted to confirm if this is the case.
>     UPDATE: I have monitored the bike with the ECU disconnected, and can confirm I am getting the same signals as when it is connected (no speedo requests). I have checked signal wire continuity, all the way from the plug under the seat to the input pins directly on the ECU circuit board and they are fine, so it isn't a wiring/plug issue.
>     I have bought a second hand ECU unit fairly cheaply, and will give that a go when it arrives.

Hi. Generally ECU for socos is the GPS unit. did you try unplugging it and see if you get display messages?

BrienAllison, 2026-10-20:
I have tried unplugging the GPS/ECU unit. Still no working speedo or battery charge info. The RS485 monitor is showing the same communication telegrams with or without plugging in the GPS/ECU unit. Only battery and controller requests/responses show. Nothing at all with the "BA" speedometer location byte.

valy84, 2026-10-20:
> 8\. I have tried unplugging the GPS/ECU unit. Still no working speedo or battery charge info. The RS485 monitor is showing the same communication telegrams with or without plugging in the GPS/ECU unit. Only battery and controller requests/responses show. Nothing at all with the "BA" speedometer location byte.

maybe you have a wiring issue on the display part. Or even a faulty 485 IC in the display. I managed to burn mine on my street hunter pro 2 years ago, but the IC is super cheap to replace, like 1..2 usd

stprograms, 2026-10-20:
I'm moving this to a discussion